kervinck/gigatron-rom #24: Make ROM loader accept zero-page segment

kervinck, 2018-06-04:
SYS_Exec_88 doesn't accept an optional (first) zero page segment, such as is allowed in GT1 files loaded over the input port. An undesirable workaround is to map them to the page at $8000, but that doesn't work in 64K systems. Can SYS_Exec_88 be enhanced? Essentially, it now does this:

```
while (hi != 0) {
  processSegment()
}
```

All that is needed is:

```
do {
  processSegment()
} while (hi != 0)
```

at67, 2018-06-09:
The following changes to SYS_Exec_88 allow for the first segment to be a zero page segment and also allow ROM files to be loaded into a 64KB RAM system without any hacks.

Tested on a real Gigatron with 64KB SRAM using tetris.rom with it's first byte reset back to 0x00.

~~~
SYS_Exec_88:  00ad 0000  ld   $00        ;  loader start address = 0x0081 - $0037
              00ae d617  st   [$17],y
              00af 0081  ld   $81
              00b0 a037  suba $37
              00b1 d21d  st   [$1d],x
              00b2 80fe  adda $fe
              00b3 c216  st   [$16]
              00b4 dc75  st   $75,[y,x++]
              00b5 dccf  st   $cf,[y,x++]        ; replace BRA with CALL, turns while{} into do{}while();
              00b6 8023  adda $23        ; new offset for call address
              00b7 de00  st   [y,x++]
              00b8 dc5e  st   $5e,[y,x++]
              00b9 dc27  st   $27,[y,x++]
              00ba dccf  st   $cf,[y,x++]
              00bb 8000  adda $00        ; offset is already at correct location
~~~

kervinck, 2018-06-09:
Thanks! I keep this open for a bit longer. I'm looking for a way to keep the ROM loader on the stack. (My "HaD" ROM did it completely wrong BTW, by putting it -under- 0x81. I didn't check why it work, but cobbling 0x80 is a bad idea].

The idea has always been that this routine can be used to page-in fragments of an active program. It should therefore put itself on at the least intrusive place in memory. Top is stack is what I came up with and hence all the bookkeeping for the relative addresses.

PS: I'm also FINALLY cleaning up the notation in theloop.py. It is slowly becoming somewhat legible.

at67, 2018-06-09:
I wondered about the 0x80 constant, I just figured the code wasn't quite long enough to cause any issues, (I never checked it myself).

Is there any reason this loader isn't just stored in video memory somewhere, (not offscreen), but onscreen and then just annihilated once the vCPU application takes over?

Also a quick question, how do you convert .gt1 files to the ROM format, manually or with some automated process?

kervinck, 2018-06-09:
In video memory it would collide with the idea than an application (such as BASIC) can use it to page-in parts of its code. It would be intrusive. Nobody uses top of stack (unless they do..).

If you look in theloop.py for 'at67' you find some commented-out code I used in the HaD ROM. To make space I further removed Jupiter from the Pictures application, modified the menu in Main.gcl and hacked SYS_Exec_88 to put the ROM loader in a different region. For that reason, Tetris.gt1 isn't the best example on how to do this :-)

kervinck, 2018-06-09:
Ok, the reason "0x81" worked is because the code subtracts two bytes too many. Older versions used this to hold a variable. Later it was changed to use sysArgs. So [0x80] was accidentally unclobbered.

`suba(53+2)                              #18 (AC -> *+0)
`

Will fix this as well.

kervinck, 2018-07-02:
Trivially solved with:
https://github.com/kervinck/gigatron-rom/commit/ca55b2c6364204f3d69e21dd2ed86c8fc8276a2e

the sub(53+2) is correct after all. The +2 is there to create stack space for the ROM loader's subroutine.

kervinck, 2018-07-02:
Tested with Contrib/at67/vCPU/graphics/lines.gt1

closed 2018-07-02 by kervinck