kervinck/gigatron-rom #39: at67 doesn't build

fuzxxl, 2018-10-15:
When I try to build at67, the build fails with the following error:

```
/home/fuz/src/repo/gigatron-rom/Contrib/at67/audio.cpp:10:10: fatal error: '../../../midi/scores/music.h' file not
      found
#include "../../../midi/scores/music.h"
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
```

I suppose the file `music.h` was accidentally removed from the project for some reason.

kervinck, 2018-10-15:
There is a file midi/music.h in this subdirectory. Perhaps this can be substituted? The branch where this came from is this one btw: https://github.com/at67/gigatron-rom
It appears to have some updates that weren't offered for inclusion yet.

kervinck, 2018-10-20:
FYI, on my macbook I just tried 
```
cd Contrib/at67
cmake .
make
```
I just get pages and pages of warnings and errors I don't know what to do with. I don't have another system at hand. I also have no idea how cmake works, to be honest.
[output.txt](https://github.com/kervinck/gigatron-rom/files/2498461/output.txt)

kervinck, 2018-10-21:
at67's fork https://github.com/at67/gigatron-rom was some commits ahead. at67 hasn't offered it as a pull request yet before falling idle, so its unclear what the status is. I can see that the music.h file is taken care of. So I just merged his fork back, with the exception of the changes in BabelFish.ino, that look rather ad-hoc and unrelated to me.

I can even get it to build on my macbook, with a small addition to CMakeLists.txt . It doesn't -work- on mac yet, but that is another matter.

fuzxxl, 2018-10-21:
The code does build now, so this seems to have fixed the problem.  I'm not sure what the problem on your machine is; perhaps try to update your C++ compiler.

closed 2018-10-21 by kervinck

kervinck, 2018-11-10:
Hi fuzxxl, I was wondering. Besides building, is the at67 tool chain also working for you? Or do we need to spend more effort in it?