kervinck/gigatron-rom #41: Fix zero page usage in Bricks and Tetronis

kervinck, 2018-11-10:
These programs try to load in the upper zero page, which causes a conflict with the ROM loader SYS_Exec_88. There is now a workaround in the menu application Main.gcl. But the workaround must go and these two applications reallocate their constants.

kervinck, 2019-01-01:
Added clarification to GT1-files.txt
https://github.com/kervinck/gigatron-rom/commit/33c090549cdb91851894f248d37cdcbbc5c1d0af

kervinck, 2019-08-10:
We might retire SYS_Exec_88 and replace it with another ROM loader.
https://github.com/kervinck/gigatron-rom/issues/84
https://github.com/kervinck/gigatron-rom/issues/46

If so, with a bit of planning we can then support loading into the zero page above $c0.

Therefore lower the priority of this specific issue.

at67, 2019-09-30:
With respect to Tetronis and Bricks, my assembler trivially supports changing the zero page _callTable_ address, (which Tetronis and Bricks currently set to 0xEE, leaving 16 bytes for vCPU stack space). The _callTable_ used by my assembler is purely for optimisation and allows vCPU code to do something like this:

CALL	clearScreen

Instead of:

LDWI	clearScreen
STW         register0
CALL	register0

So whilst it would be trivial to change the 0xEE address to something below 0xC0 and reassemble the code for both Tetronis and Bricks, they both use most of the zero page for variable space and it would probably be impossible to just move _callTable_ to below 0xC0.

The other solution is to disable the Assembler’s _callTable_ functionality, (which is trivial and doable now), and modify the original source code for both Tetronis and Bricks to replace the CALL’s with the 3 instruction group shown above. The problem with this solution is that both Tetronis and Bricks code and data are tightly packed within the lower RAM and video pages and blindly replacing CALL’s with the above 3 instructions would definitely cause page misalignment issues and break the code in a myriad of ways.

To resolve this problem completely would require both applications to be re-written with the new 3 instruction alignment caveat in mind and would be non trivial at best.

The BASIC compiler I started writing last year does not suffer from this issue, it generates vCPU code using the 3 instruction paradigm for CALL’s shown above.

kervinck, 2019-09-30:
Ok thanks for the analysis! The issue is bigger for the SDC/MMC loader, and on the long run that one is more important. So I want to focus on making these programs load without modifications. Hence I labeled this as "low" priority.

One idea I have is to park the page 0 segment elsewhere during loading, and copy it back before execution. On a 32K system, the videoTable comes to mind.

PS1: the recently proposed CALLI $HHLL instruction might be useful to reduce the pressure on the zero page for larger programs. This also helps the SDC/MMC loader itself. And besides reducing zero page pressure, it makes it easier to deal with segmentation of code blocks.

PS2: I haven't been able to experience your assembler yet. Your recent edition of gtemuAT67 comes up like this on my MacBook Pro, macOS 10.14.6 ("Mojave"):

[image]

kervinck, 2019-09-30:
```
LDWI	clearScreen
STW register0
CALL	register0
```

You can already replace that with

```
LDWI clearScreen
CALL  vAC
```

The advantage of a CALLI $HHLL instruction will be that it takes 3 bytes instead of 5, and that it doesn't clobber vAC.

at67, 2019-09-30:
I can't see an issue with saving and restoring page 0 during loading, assuming you can afford the ROM space for the additional code, it sounds like a robust solution.

The CALLI $HHLL instruction will be a welcome addition and I have no idea how I missed the CALL vAC instruction, I don't think I even included it in my assembler; I'll have to check!

Apart from the emulator booting up with ROMv1, what you have there seems fairly normal, does it run ok? The new version that I will upload shortly supports multiple internal and external ROM's that you can switch between at your leisure.

kervinck, 2019-09-30:
I will describe my macOS experiences with gtemuAT67 in a new issue #114, if you don't mind

kervinck, 2019-10-02:
Close issue. The path forward is more flexible loading capabilities.

kervinck, 2019-10-06:
Reading your analysis again, it occurs to me there may be an understanding. The issue is not about _using_ $c0 - $ff. The issue is preloading that area with values. If the jump address table can be moved away from $c0 - $ff, then that area becomes available for the program's variables again.

Also the SDC/MMC loader will need quite a bit zero page space to operate. If I can constrain that to $c0-$ff, than that is the best of all worlds. (No need to relocate stuff just because of these two programs)

at67, 2019-10-07:
Ok, apparently I never fully understood that the space above 0xC0 was only off limits during loading and that after loading had completed it was completely free to use as the programmer sees fit.

Given that, I have modified everything in the Contrib section that caused page0 issues, I haven't made a pull request or even checked it in yet as I would like to get your thoughts on some of the changes I have made.

Everything that I modified to not trample on 0xC0 onwards works in emulation and on real hardware, (I double checked it all), Xbx's Bricks was the one application that gave me the most grief as it actually has 2 page 0 segments and would 100% crash or be naughty if I tried to move those segments anywhere other than where they were originally created.

My assembler merges multiple page 0 segments into one page 0 segment, but it turns out that there was a nasty bug in this code that only manifested itself for Bricks when trying to relocate it's multiple page0 segments.

Anyway here is a list of the changes in terms of files touched, any source code that I modified that wasn't mine I left the original version with an _old extension, we should probably try and contact the original authors to let them know of the changes, (only Xbx so far I think). All the vCPU changes were just minor relocation values that needed to be changed, once I fixed the page 0 merger.

Also some of Xbx's code was not buildable as his macros folder was missing, I think he had a symbolic link on his system into a test area with a copy of my macros files, I placed a proper copy of the files into his Contrib area and now all his code is buildable again.

Here is a list of the files I had to touch:

[image]

kervinck, 2019-10-07:
Thanks! That's a big change. The irony is that when I wrote SYS_Exec_88 for ROM v1, I decided that it needed put its code on the least intrusive place in memory. So that became on the user stack. The idea has always been that SYS_Exec_88 can also be used to page in code segments from RAM, like loading fragments from a library. Or like switching between different parts of the same application. With that in mind, it should never ever thread on memory that might be used by the application. So putting the code on the stack seemed perfect. Oh irony...

I will be very busy until after next weekend. We're not in a hurry here.

at67, 2019-10-07:
No worries, I'll push it into my area for now and do a pull request when you have time

kervinck, 2019-10-08:
Thanks. Writing down the intent of interface.json the Docs/GT1-files.txt helped me a lot to think this through. It works out reasonably well, with the occasional headaches that come with it.

For my understanding, I created some heat maps out of the existing GT1 files.

[image]

[image]

at67, 2019-10-08:
Yeah the 64k midi demo has about 32k of midi files in it that play for about 30mins, some of my favourite old-school 8 and 16bit tracks.

It's enlightening to see a statistical analysis visually like this, I did something similar with native instructions based on dhkolf's getemu optimisations, (where he hard coded the 3 most common instructions).

I took it a step further and ran multiple gt1 files for extended periods of time whilst collecting instruction usages stats, (the code for it is simple and checked in), and then optimised all instructions, (in a similar way to dhkolf), that had 1% or more usage.

It only made around a 5-6% gain in my emulator's performance, (I was actually expecting lower as my emulator spends the majority of it's time pushing pixels to a screen buffer and doing debugging etc, (I don't use any of the SDL API for any of the graphics, it's all pixel/byte poking like the good old days. 

But it was eye opening and a lot of fun!

kervinck, 2019-10-08:
I wonder how hard it is go visualise the program counter on a 800x521 grid as the beam scans across. For example, plot PC8:13 as a color. Or alternatively something more elaborate. "The last PC0:5 target since the most recent "jmp Y,xxx" statement, or any backwards branching instruction". It should be possible to point at the different Gigatron functions. When it is doing sound, when vCPU, when serial input etc...

at67, 2019-10-08:
Yeah that would be very cool, for both native and vCPU PC's, a real time memory heat map and program counter heat map would be neat.

One thing I have also been doing with my emulator the last couple of days is just single stepping through vCPU code and native code simultaneously. 

My debugger now allows you to switch anytime whilst debugging any code between vCPU and native, it's great single stepping on the macro level in vCPU and then dive bombing into the native code at the micro level and watching just how much work the Gigatron really is doing.

at67, 2019-10-11:
No need to reply to this until you have time:

Another issue I found with moving page0 segments around is that the loader uses 0x30 to 0x44 for it's variable space. As you can probably guess some of the samples, (especially after the changes I made), now define constants in this area so as to try and alleviate the pressure on the >=0xc0 area.

I've added more code to the assembler and loader that treats each entry within a call table as a separate entity and then merges them all together just before the gt1 file is constructed. This allows the call table to be placed anywhere in page0 and in any order, (even around 0x80 for example), this negates the need for a contiguous section of page0 for the call table and allows stuff to basically go anywhere.

Tetronis is the closest to breaking, it uses constants starting at 0x4a and finishes with variables at 0xE5, (constants finish well before 0xC0), thus leaving a small buffer for Loader's variables to grow in the future and just enough room for the nested call stack required for Tetronis to run.

I think sometime in the near future this will need to be documented, obviously GCL and BASIC aren't/shouldn't be affected by this at all, but vCPU stuff with it's freedom of variable/constant placement will need strict rules in place to stop potential problems. Some or all of these rules could be built into the Assembler, but until we know exactly where and what size these page0 limits are, (will they be variable between ROM versions?), then I will hold off on any more code in this area.

kervinck, 2019-10-14:
Argh yes, I overlooked this... I think it's wise we shouldn't change anything a bit later down the road it becomes fully clear what we really _need_ and what we really _can do_ with the SDC/MMC loader.

at67, 2019-10-14:
Would you like me to do a pull request with what I currently have? All the code samples are compatible with all versions of ROM and all work on real hardware, (and meet the requirements of not munching on Loader's variable space or using constants past 0xBF).

Also what would you like me to do with Xbx's original code, leave it as is, (renamed to _old), or remove it from his Contrib section?

kervinck, 2019-10-17:
Solved in pull request #119

closed 2019-10-17 by kervinck

kervinck, 2019-10-26:
Wow, the Tetronis music sounds a lot better as well.

at67, 2019-10-28:
I optimised the MIDI player a bit and spent some time making sure the audio was being processed often enough in full scanline mode; it works pretty well now, 

I also updated the GCL and VASM MIDI examples the same way and added code for displaying an error when trying to run 64K examples on 32K systems.

kervinck, 2019-11-10:
BTW, when I used your MIDI player for Apps/Overworld.gcl, I spotted this bug in your GCL versions:

```
{ Loop forever }
[do
  $10 \soundTimer=                          { keep pumping soundTimer, so that global sound stays alive }
```

This overwrites 2 bytes and stops the LED pattern. I suspect you want this:

`\soundTimer.
`

at67, 2020-01-28:
Unngh, no idea how I missed this comment, thanks for spotting this long standing bug!

Finally fixed.