kervinck/gigatron-rom #59: Building latest Contrib/at67

zborac, 2019-04-20:
I have successfully built Contrib/at67 (from latest commit 4cd5539  on Jul 28, 2018) using gcc/mingw-w64 (gcc versions 7.2.0, mingw-w64...win32-seh) and "Visual Studio 2019 community" both. SDL2 version is 2.0.9. Target architecture is x86_64 in both cases.
Everything I can get in console window, when start this emulator is perpetual:
	main(): CPU stall for 12500004 clocks : rebooting.
	...
This happens either executable was built using gcc/mingw-w64 or "Visual Studio 2019 community".

kervinck, 2019-04-20:
Hello Zborac.

There has been a thread in the user forum with additional build instructions: https://forum.gigatron.io/viewtopic.php?p=368#p368

If that doesn't help, I suggest you open a new thread in the user forum, so the community might be of help. In this case, I'm not capable enough to support Windows.

zborac, 2019-04-20:
Thanks. Very helpful. The issue has been resolved.

kervinck, 2019-04-20:
Good to hear!

closed 2019-04-20 by kervinck