kervinck/gigatron-rom #68: lcc: add 0x81-0xbf to zero page segment

kervinck, 2019-05-09:
Globals and static variables in functions are (much) faster when they can be placed in the zero page rather than main memory. We can give this zero page real estate to as many as such variables as possible. The call stack doesn't need much space after all.

Perhaps let the programmer hint which variables are candidate for such preferential treatment. ("_static register_" comes to mind, but that's not a fully developed idea, because we can have pointers to such variables.)

kervinck, 2019-05-13:
Low priority

lb3361, 2022-08-25:
This issue is still relevant to glcc since it does not place globals in page zero.

This would require decorating the global declaration with attributes requesting a page zero placement for a global. Since the code generator already supports such variables, and since glink already tracks the page zero usage, the main difficulties are (1) adding an attribute syntax for lcc declarations, and (2) making sure that the linker does not mix zp globals and normal globals.

lb3361, 2023-07-18:
glcc can now place globals in page zero using the __near type qualifier.