kervinck/gigatron-rom #69: lcc: complete vnsprintf

kervinck, 2019-05-09:
1. Handle EOF error from putc
2. Handle size, flags, width and precision
3. Implement %o %x %X %p
4. Cross-check with C89 standard instead of K&R2
5. Make charout() _much_ more efficient with a different strategy to detect streams. Mark stream functions by passing a negative (int)size to vsnprintf. Then the stream/string decision can be made once at the top of vsnprintf. Streams then continue writing as a string in their buffer. (If the buffer is full, flush it and continue.)

kervinck, 2019-05-09:
On 5, we can use a more subtle mechanism, because UINT_MAX is not an unreasonable value for 'size' in real code.. A new idea is to flag streams as (str+size == 0) or (str+size == 1).

kervinck, 2019-05-13:
Step forward: https://github.com/kervinck/gigatron-rom/commit/0fea8bbab692db8b5d1ba6265956523dd909c9e9

lb3361, 2022-08-25:
Suggesting to close this issue since glcc has this already.