kervinck/gigatron-rom #70: lcc: optimisations

kervinck, 2019-05-09:
[Note: This issue is an aggregation for optimisations of the emitted code. I want to park all ideas here for future reference. Regarding priorities in LCC the order should be 1. Correctness, 2. Usability, 3. Optimisations.]

Ideas (some simpler than others. some realistic, some are nonsense):
1. POKE is often preceded by ANDI 0xff, but this is almost never needed
2. entermask/leavemask can sometimes use LDI
3. many cases of stw(x) + ldw(x) or other way around. Can be optimised
4. eliminate ldw(vAC) and stw(vAC)
5. if we have a known value in vAC, use SUBI to get a small negative number
6. comparisons eq/ne with small negative constants can avoid LDWI
7. don't 'pusha' each argument, but allocate the argument area in one go
8. option to use vCPU stack as data stack (lots of work, unclear if it will bring anything)
9. use DEF for string pointer initialisers
10. is it feasible to use INC more often?

kervinck, 2019-05-13:
Low priority

kervinck, 2019-05-15:
Idea to reduce the number of @thunk functions in rt.py:

1. Move the start of the pixel lines to offset 0x60.
```videoTable[1] = 0x60;```.
Pixels then live from 0x60 to 0xff and code can live at offset 0. This eliminates the need for @thunk1

2. @thunk2 hops over from the end of page 4 into page 8. We can do that by placing @thunk2 at the beginning of page 5 (stack area), and move there with thunk0 from page 4.

```
STW  tt
LDW  0x800
STW    vLR
LDW  tt
RET
```
In essence, this eliminates a zero page reference.

kervinck, 2019-05-17:
13. Use XORW instead of SUBW before the == and != operators.

kervinck, 2019-05-18:
14. MULI2(CNSTI2(1), ...

Simplify. Same for CNSTI2(0) and MULU2, DIVXX etc

kervinck, 2019-05-18:
15. Remove 'rv' from rt.py, use 'ha' instead for return values

kervinck, 2019-06-01:
16. More aggressive purging of unused library functions. For example:

```
int main(void)
{
  return 0;
}
```

Still gives an .gt1 file of more than 4 KB in size. It seems that references from other functions are not purged (e.g. _div_ in rt.py references _divu_, and _divu_ is never purged because of this?) Also some references come from the data spac, such as the _flush_ methods in the FILE objects of stdin.c/stdout.c).

kervinck, 2019-06-01:
17. LCC inserts explicit return values in places where "don't care" will work. This makes some code larger than needed.

See this comment to #76 for an example: https://github.com/kervinck/gigatron-rom/issues/76#issuecomment-497897449

kervinck, 2019-08-27:
18. Some C11 code crept into src/gt1.md. Workaround solved with #97. It's would be nicer to make it all ANSI C compliant (aka C89).

lb3361, 2022-08-25:
This issue should be closed since glcc already does such optimizations (when they make sense in the new code)