kervinck/gigatron-rom #73: lcc: compiler crash

kervinck, 2019-05-11:
```
$ cat test2.c
struct S {
  char c;                       // 'int c' works...
};
int f(int c, struct S *s)
{
  return s->c = c;              // just 's->c = c' without return works...
}
$ make test2.gt1
Utils/lcc/build/lcc -ILibs -c test2.c -o test2.o
Utils/lcc/build/lcc: fatal error in Utils/lcc/build/rcc
make: *** [test2.o] Error 1
$
```

kervinck, 2019-05-13:
Crash happens in emitasm, before returning:

`emitasm(0x7fc2a3012338, ASGNU1, (none)) (action)
`

kervinck, 2019-05-13:
Simplification of the test program:

```
char x;
int f(int c)
{
  return x = c;
}
```

kervinck, 2019-05-13:
Simplification of test program:

```
char x;
int f(int c)
{
  return x = c;
}
```

kervinck, 2019-05-13:
This was commented out in https://github.com/kervinck/gigatron-rom/commit/e58bf0c1c3e4b975875aedfb65f0398835d8d59e

```
//	case CVI: case CVU: case CVP:
//		if (optype(p->op) != F
//		&&  opsize(p->op) <= p->syms[0]->u.c.v.i)
//			p->op = LOAD + opkind(p->op);
//		break;
```

If we re-enable, the compilation completes without crashing LCC itself. Resulting output:

```
asm.defun('_f')
asm.push()
asm.ldwi(0xc0)
asm.call('enter')
asm.ldi(6)
asm.call('ldloc')
asm.stw('r7')
asm.ldwi('_x')
asm.stw('r6')
asm.ldw('r7')
asm.poke('r6')
asm.ldw('r7')
asm.stw('rv')
asm.label('.L1')
asm.ldwi(0x600)
asm.call('leave')
asm.ldi(2)
asm.addw('sp')
asm.stw('sp')
asm.ldw('rv')
asm.pop()
asm.ret()
asm.defun('_x')
asm.dx([0] * 1)
```

kervinck, 2019-05-13:
This would be a good point to ask @pgavlin for some insights :-)

pgavlin, 2019-05-13:
This is the point at which I wish that I had gone back and commented things up a bit more.

I disabled the conversion of CVI/CVU/CVP to LOAD because it was breaking the backend's assumption that all LOADs are from vAC to a virtual register or vice-versa. We could probably tweak things such that the conversion works, but I'm not 100% certain what those tweaks would be offhand.

It's a bit strange that the register allocator appears not to be allocating those nodes... I'll try running `cpp` and `rcc` manually (rather than via the driver) to get debug output.

kervinck, 2019-05-13:
I can work-around it in my C library and made a note in my updated Libs/stdio/fputc.c that found the issue.

kervinck, 2019-05-19:
Fyi, I spent the best part of the last few days on the other issue:
https://github.com/kervinck/gigatron-rom/issues/76 <https://github.com/kervinck/gigatron-rom/issues/76>

Basically: nested multiplicative expressions such as "a * (b / c)" don’t compile.

LCC gets in an infinite loop of spilling and reloading vAC but I’m stuck at understanding it.
It spills vAC at some point, which is fine and understandable.
Later it reloads, but at that point vAC is allocated, so it wants to spill it again. etc

(The message “inserting load from vAC to vAC” looks like a red herring.)

lb3361, 2022-08-25:
This issue should be closed since glcc fixes it.