kervinck/gigatron-rom #77: lcc: compiler hangs on simple statement

kervinck, 2019-05-27:
```
$ cat hang2.c
int f(int c)
{ 
  if (c == -1)
    return 0;
}
$ make hang2.gt1
Utils/lcc/build/lcc -ILibs -c hang2.c -o hang2.o
hang2.c:5: warning: missing return value
Assertion failed: (offset < 0x10000lu), function local, file ./build/gt1.c, line 4435.
Utils/lcc/build/lcc: fatal error in Utils/lcc/build/rcc
make: *** [hang2.o] Error 1
$
```

Similar to #76 but not the same.

kervinck, 2019-06-04:
This is in fact the same type of register spilling issue as #76. We can solve it with #76, or workaround it by replacing SUBW with XORW for the relational operator (this could be part of #64).

lb3361, 2022-08-25:
This issue should be closed since glcc fixes it.