kervinck/gigatron-rom #89: Limited Program Memory for BabelFish in ATTiny85

ryanohoro, 2019-08-11:
The current GT1 files loaded on the ATTiny85 (flash size < 10,000) includes TinyBASIC and leaves only 156 bytes of PROGMEM left for new feature development.

Since TinyBASIC is included in ROMv3, it should now only be loaded in larger devices (flash size >= 10,000).

This change would require Pluggy McPlugface users with < ROMv3 to remain on an earlier version of BabelFish for TinyBASIC.

kervinck, 2019-08-12:
Hi @ryanohoro, could you elaborate: is this a duplicate of #90 or does it serve another purpose? Removing TinyBASIC is the easiest method to free up space for new features. Back in the day I spent quite a bit of time to compress the TinyBASIC. object file and win something like 300 bytes. But in the end I decided it's not worth it to add all that complexity. (In a more reasonable development, files will be stored on a memory card connected to the SPI expander)

ryanohoro, 2019-08-12:
That sounds great!

This is not a duplicate. A pull request was opened with a code change to help resolve this issue, but I have closed it for now.

I understand you're also busy. If you'd like, I can finish any work you may think is appropriate to free up space, and do so in my own fork.

kervinck, 2019-08-13:
Another suggestion is is to do the LED work in Contrib/rynahoro/
It's then clearly visible for everyone interested, and we can always decide at which point to adapt it to the kit edition.
There is already code in BabelFish that sends commands _to_ the keyboard.

closed 2019-08-24 by kervinck