kervinck/gigatron-rom #91: lcc doc: supported/unsupported functions unclear

jurkstas, 2019-08-12:
It is not completely clear to me as a user what lcc functions are supported (implemented, working? not sure about the proper word here) and what are not. https://github.com/kervinck/gigatron-rom/blob/master/Utils/lcc/README.md lists some examples of what is working and what is not, but would it be possible to have a list of implemented/not implemented features as part of documentation? Even if it is incomplete.

How could someone help?

kervinck, 2019-08-12:
This is an open-ended question that is hard to answer. If you can make the question specific, it shouldn't be too difficult with yes/no. I fear the original author has moved on to other projects.

kervinck, 2019-08-12:
About what works: there's an example in Libs/Example.c, Libs/*/*.c and Utils/lcc/gt1/none/tst/*.c 

About what doesn't work: The known non-working parts are covered by the GitHub issues with 'lcc' in the title.

kervinck, 2019-08-12:
The README.md file is indeed as good as I can summarise it

closed 2019-08-24 by kervinck

kervinck, 2019-12-07:
commit c80bbc1f623572657c3a1206ef558045db51e538
Author: Marcel van Kervinck <marcelk@bitpit.net>
Date:   Sun Oct 6 19:32:20 2019 +0200

    lcc: Explain limitations better in README.md

commit f317e35ada1c49c96cd60c91b76d73391a6caa24
Author: Marcel van Kervinck <marcelk@bitpit.net>
Date:   Sun Oct 6 19:29:07 2019 +0200

    lcc: Explain limitations better in README.md

commit 7288fe4f19e4f38ed46d5a58234f36110bdf1c9b
Author: Marcel van Kervinck <marcelk@bitpit.net>
Date:   Sun Oct 6 18:58:14 2019 +0200

    lcc: Explain limitations better in README.md