kervinck/gigatron-rom #121: Migrate from Python 2 to 3

kervinck, 2019-10-25:
```
./Core/compilegcl.py:#!/usr/bin/env python
./Core/gcl0x.py:#!/usr/bin/env python
./Core/ROMv3y.py:#!/usr/bin/env python
./Core/ROMv3.py:#!/usr/bin/env python
./Core/ROMv2.py:#!/usr/bin/env python
./Core/ROMv1.py:#!/usr/bin/env python
./Core/ROMv4.py:#!/usr/bin/env python
./Core/dev.py:#!/usr/bin/env python
./Utils/raw24toraw6.py:#!/usr/bin/env python
./Utils/sendFile.py:#!/usr/bin/env python
```

TeddyBear06, 2019-11-17:
Hello,

I would like to contribute on this issue.

Are you in a hurry ? I mean, there is a short deadline ?

I already "fixed" Core/asm.py but I need a little more time to finish all files.

kervinck, 2019-11-17:
Oh that would be great. Be my guest!

The Python project itself stops supporting Python2 from January 1st, 2020. So I was going to do this with xmas anyway.

TeddyBear06, 2019-11-17:
Thanks for your welcoming words.

I'm currently converting strings bindings in asm.py :-)

TeddyBear06, 2019-11-18:
Some more interesting concerns found so far :

1. `int` and `str` comparison is not possible anymore [like here](https://github.com/kervinck/gigatron-rom/blob/master/Core/gcl0x.py#L160)

```python
if 0 <= con < 256:
```
> Where `con` can be 3, None or 'Setup'.

It should become something like :

```python
if isinstance(con, int) and 0 <= con < 256:
```

2. We cannot write strings directly in files opened in binary mode anymore [like here](https://github.com/kervinck/gigatron-rom/blob/master/Core/compilegcl.py#L92).

```python
TypeError: a bytes-like object is required, not 'str'
```

But I cannot figure out what encoding we need...

**Side note** : I used for the first time the [`2to3`](https://docs.python.org/3.9/library/2to3.html) tool from Python and it works pretty well! It fix all missing parenthesis of Core/asm.py for example, just by running :

```shell
2to3 -w Core/asm.py
```

kervinck, 2019-11-19:
Ah yes, there are always these little surprises popping up.

I'm glad we normally check-in all .rom and .gt1 files, also older releases. That makes migrations like these very well testable: `git diff` shouldn't flag any such files.

PS: I'll be busy for the rest of the week as I must push my Gigatron activities more into the weekends.

TeddyBear06, 2019-11-19:
Yes, I noticed the usefull result of git diff after a make compiletest yesterday. Ideal for this kind of migration.

No problem, I'll try to finish until the end of this week.

TeddyBear06, 2019-11-24:
Hello,

I'm sorry to inform you that I didn't found a way to manage .gt1 encoding correctly during `make compiletest` (`git diff` shows modifications).

If I'm not mistaken, right now we handle [raw bytes in a string](https://github.com/kervinck/gigatron-rom/blob/master/Core/compilegcl.py#L53) that we [write directly to a file using binary mode](https://github.com/kervinck/gigatron-rom/blob/master/Core/compilegcl.py#L91).

In Python 2, all encoding weird stuff was done implicitly (as open uses locale.getpreferredencoding() it can be different on Windows or Linux).

In Python 3, byte is a specific type. If we open a file in binary mode, it expects to receive bytes-like object to write succesfully to the file. 

I tried all possible encodings on all .gcl files and found that sometimes ascii works (especially for Blinky.gt1 and SmallTest.gt1) but fails for others...

I'm stuck.

kervinck, 2019-11-24:
Ah yes. I have to dig into how this works again. I recall asm.py uses two plain string to remember bytes. One for even bytes and one for odd bytes. This originates from the breadboard version, which had two ROM chips for that. Maybe I have to through away that and construct it again using p3 building blocks. We have plenty of time, thanks for helping out. Give it a rest! It is supposed to be fun!

TeddyBear06, 2019-11-24:
Sure! I love this project!

psr, 2019-11-27:
I'm sorry if PR #129 stamps on your work on this so far @TeddyBear06 (if it's merged) - I wanted to make some changes in asm.py, and didn't want to deal with a difficult merge later, so I wanted to get that file at least ported. To work out if I'd done it right I needed to port a few other files too. I don't intend to do any more of them.

TeddyBear06, 2019-11-27:
Hi @psr,

Sorry for the delay. 

No problem! 

If I'm not mistaken you did manage byte handling in asm.py! Wonderful :-)

I can rebase my "modifications" on yours without any problem.

Thanks for letting me know about that.

kervinck, 2019-12-25:
The transition should be complete now. Bye bye Python 2!

closed 2019-12-25 by kervinck

psr, 2019-12-26:
Do you have a minimum version of Python 3 in mind? I'm currently developing my Forth against 3.8, but if there's a standard, I don't think it would be hard to change.

kervinck, 2019-12-27:
I'm not deep into it. Just what is commonly available on all platforms I suppose?

psr, 2019-12-27:
I think may projects try to be compatible with at least 3.6, as that's installed by default in the current Ubuntu LTS (I believe). I've just tried my code with 3.6, and the tests pass, so it must be fine 🤞

kervinck, 2019-12-27:
Ah good to know. The nice thing about Python 2.7 was that we didn't have such headaches. Platforms tend to be conservative. I recall Utils/lcc needs 3.6, and that was a bit of an issue when LCC arrived earlier this year. But it should be fine now indeed.