kervinck/gigatron-rom #135: Win10, Python 3.8.1 error, when running compilegcl.py

at67, 2019-12-27:
compilegcl.py returns the following error when trying to apply the ROM v1 Loader patch to a *.GCL file on Win10 using Python 3.8.1
```
Apply patch $5b86
Traceback (most recent call last):
  File "Core\compilegcl.py", line 72, in <module>
    data += ''.join(chr(byte) for byte in [
TypeError: can't concat str to bytearray
```
I changed the relevant code from this:
```
  data += ''.join(chr(byte) for byte in [
    patchArea>>8, patchArea&255, 6,   # Patch segment, 6 bytes at $5b80
    0x11, address&255, address>>8,    # LDWI address
    0x2b, 0x1a,                       # STW  vLR
    0xff,                             # RET

    0x00
  ])
```
To this:
```
data += bytes([
    patchArea>>8, patchArea&255, 6,   # Patch segment, 6 bytes at $5b80
    0x11, address&255, address>>8,    # LDWI address
    0x2b, 0x1a,                       # STW  vLR
    0xff,                             # RET

    0x00
  ])
```
And it works correctly now, tested under emulation and on real hardware.

kervinck, 2019-12-27:
A great! I did a 'make compiltest' before committing the Python 2 to 3 but missed this. Thanks for spotting!

kervinck, 2019-12-27:
Merged and retested ok!

closed 2019-12-27 by kervinck