kervinck/gigatron-rom #161: GTOS: Vol.ID 0000 Failed

kervinck, 2020-02-10:
Finding the partition doesn't work for all SD cards. It seems the wrong sector is read.

https://forum.gigatron.io/viewtopic.php?p=1215#p1215
https://forum.gigatron.io/viewtopic.php?p=1334#p1334

kervinck, 2020-02-10:
From skimming over the older [I]CardTest[/i] code, I conclude that byte addressing isn't detected correctly. And even it if were, it wouldn't work anyway, because the way it is handled further on isn't correct. I believe this is consistent with the observations so far, including [url=https://forum.gigatron.io/posting.php?mode=edit&f=4&p=1334]jwolfram's[/url] today. This is purely a software issue to be solved in GTOS at some point perhaps. It isn't related to the expander circuit itself.

The first fix is to reject byte addressing cards, instead of failing later on.

lb3361, 2022-08-25:
Suggesting to close this issue since this has been fixed in CardBoot (but was trickier than expected: https://forum.gigatron.io/viewtopic.php?t=369)