kervinck/gigatron-rom #163: Security issue Contrib/PhilThomas/package-lock.json ("minimist")

kervinck, 2020-03-15:
I have another warning from Github about the Gigatron repo:

[image]

PhilThomas: Would you mind to take a look?

It would be worthwhile to find out how we can avoid getting these security issues effecting the Gigatron repository. I fear one day GitHub might lock us out because of these things. Any suggestions are welcome.

PhilThomas, 2020-03-15:
> issues affecting the Gigatron repository. I fear one day GitHub might lock

I'm on it.

Phil

kervinck, 2020-03-15:
The security banner disappear after merging https://github.com/kervinck/gigatron-rom/pull/164
Thanks for the super fast response. These things are nerve wrecking to me.

closed 2020-03-15 by kervinck

PhilThomas, 2020-03-15:
No problem.  It was a dev dependency, not a deployment dependency, but
there are no longer any dependencies of either kind, so this should not
happen again.

Cheers

Phil