kervinck/gigatron-rom #182: gtemuAT67: single " crashes tokenizer(?)

jurkstas, 2021-01-06:
When I try to load a .gbas file, Exception is thrown:
```
Exception thrown: read access violation.
**str** was 0x992CB00000.
```

Crashing code:
```
_runtimePath_ "../runtime"
_runtimeStart_ &hABCD"
```

Notice that extra `"` at the end.
Debugger stops at this line:
https://github.com/kervinck/gigatron-rom/blob/af4e780a3141eb3d3ef9f0f6f39b9b301ba495fb/Contrib/at67/expression.cpp#L933

at67, 2021-01-07:
Cheers for the bug report:

This has already been fixed in the current internal version 1.06R, the code was changed from:
~~~
while((*str  &&  *str != c)  ||  (numQuotes & 1)) 
~~~
To:
~~~
while(*str  &&  (*str != c  ||  (numQuotes & 1)))
~~~
The bugged code could read past the end of the string trying to access invalid memory, if there was an odd number of double quotes or a trailing double quote. The new code changes the while check to not allow the numQuotes bool to circumvent the delimiting zero check.

I'll leave this issue open until the fix in 1.06R has been verified when it is released.

*Note* The trailing double quote is now silently ignored, extra error reporting code needs to be added.

at67, 2021-01-09:
So far this issue seems fixed, so I am going to close it here; if it crops up again, create a new issue at https://github.com/at67/gigatron-rom/issues.

Cheers.

closed 2021-01-09 by at67