kervinck/gigatron-rom #192: Comparison bug in tinybasic?

lb3361, 2021-04-15:
I would have expected the last statement below to print nothing (not "yes")

[image]

lb3361, 2021-04-15:
Possible fix:
```
diff --git a/Apps/TinyBASIC/TinyBASIC_v4.gcl b/Apps/TinyBASIC/TinyBASIC_v4.gcl
index 9b3e497d..80144cbd 100644
--- a/Apps/TinyBASIC/TinyBASIC_v4.gcl
+++ b/Apps/TinyBASIC/TinyBASIC_v4.gcl
@@ -452,7 +452,7 @@ GetLine=
     Expression!
     Keyword! `then #0           {Optional}
     %2 Value^                   {Do the comparison, beware of overflows}
-    [if<0 %2                    {Opposite signs}
+    [if<0 %2 1|                 {Opposite signs}
      else %2 Value-]            {Same signs}
     [if>0 4]                    {First is greater}
     [if<0 1]                    {Second is greater}
```

at67, 2021-04-27:
You can close this issue at your leisure, thanks for fixing it!

closed 2021-04-28 by lb3361