kervinck/gigatron-rom #205: Gigatron does not return when enteriing SAVE in MS BASIC

0ric1, 2021-09-04:
I have a problem that the Gigatron does not return after entering SAVE in MS BASIC,
I could save and load in TinyBasic but SAVE in MS BASIC does not work - it shows Saving but then the prompt does not came back,, I could load the program saved with TinyBasic in MS BASIC.
Thanks for the help.

at67, 2021-09-18:
Looking at MSBASIC_v1.gcl it seems Marcel implemented SAVE to BabelFish in 6502 assembly using this routine:
~~~
{-----------------------------------------------------------------------+
|       SAVE                                                            |
+-----------------------------------------------------------------------}
*=$3000

_Buffer=$2405

_SAVE=*         #_PHP_                  {Preserve parsing state}
                #_PHA_
                #_LDYIM_ #0             {Print message}
_save1=*        #_LDAAY_ ##_SaveMsg
                #_INY_
                #_JSR_   ##_CHROUT
                #_BNE_   #@_save1
                #_LDAIM_ #<\Buffer      {Serial out buffer}
                #_LDYIM_ #>\Buffer
                #_STAZ_  #<i
                #_STYZ_  #>i
                #_LDAIM_ #0             {Bit counter}
                #_STAZ_  #<j
                #_LDAIM_ #10            {Previous char}
                #_STAZ_  #>j
                #_LDAIM_ #<\SaveChar    {Redirect BRK}
                #_LDYIM_ #>\SaveChar
                #_STAZ_  #_vPCL
                #_STYZ_  #_vPCH
                #_PLA_                  {Restore parsing state}
                #_PLP_
                #_JMP_   ##_LIST        {Continue as LIST command}

_SaveMsg=*      #13 `SENDING`TO`BABELFISH #13 #0
~~~
Does this message, "SENDING TO BABELFISH", appear when you type SAVE?

0ric1, 2021-09-18:
The message appears but the screen goes blank, so I can't really ses what the message is - only recognized sending ...

at67, 2021-09-18:
I can't verify TinyBasic or MSBASIC's LOAD, (CTRL-F3), or SAVE mechanism on my hardware as I am running on a modified BabelFish, (which I will eventually have to fix); a couple of things you can try:

1) SAVE modulates vertical sync to output across the serial port, this can upset some monitors. Have you tried giving it 30 seconds or so, (SAVE is very very slow, about 6 bytes per second), and then power cycling your monitor to see if the video signal has been restored?

2) Start a thread here https://forum.gigatron.io/viewforum.php?f=4 and see if anyone else has the same issue, there are plenty of other Gigatron Hardware owners with Pluggy who could verify if a bug has crept into BabelFish.

tgschultz, 2021-12-13:
I've been writing a Gigatron emulator as a side project and ran across what I believe to be this same issue. Naturally I first thought it was an issue with my implementation of the saving protocol (which works fine for BASIC), but some debugging suggests the problem is actually on the Gigatron ROM end. 

The vblank hcycle count is always 8 during the MSBASIC save on ROMv5a and the save routine seems to loop forever. If the MSBASIC program is  >2 lines there will be visible rasterization issues as well. As I am not well versed in 6502 assembly nor do I have a familiarity with the Gigatron ROM VM internals, I have yet to debug further.

at67, 2021-12-17:
Thanks for your input, when I get a chance these holidays I'll take a look at it and see if I can spot anything.

lb3361, 2023-11-12:
I believe the following patch in MSBASIC.gcl fixes the problem:
[code]
 +-----------------------------------------------------------------------}
 *=$3100

-_SaveChar=$30fe                 {vCPU adds 2 and wraps to $3000}
+_SaveChar=$31fe                 {vCPU adds 2 and wraps to $3100}
 [do
   {
     Alternative "MainLoop" for when we're in SAVE. This catches the LIST
[/code]
I attach the modified gt1 file for MSBASIC which can be executed from the loader.
[MSBASIC_vX.zip](https://github.com/kervinck/gigatron-rom/files/13329382/MSBASIC_vX.zip)