kervinck/gigatron-rom #263: 74HC version

vslinuxdotnet, 2024-09-25:
Hello,
Congrats for the project!
I have some old 74hc logic chips and i just order the board to and build the project, but i can't get it working.

Questions:
Can run with all chips 74HC logic ?
Can run at 6MHz or 8MHz (don't have a 6.25MHz)?
Works with M27C1024-20FI EPROM?
Can i do a manual reset  by short to ground the reset pin (U6) ?

Symptoms:
The four leds at power on time: blinks all, on all, off all, some on, some off, just random.

What I do :
Checked the clock with oscilloscope and i get the clock.
They pull 205ma from 5v/3Amp power supply.
I checked all the Ic's in TL866+ and all is OK.
checked all the connections in IC's.

what can i can do to debug it more ?
Thanks

abcabce123123, 2024-09-25:
This is close! Please!!

vslinuxdotnet, 2024-09-25:
> […](#)

sorry, not the best subject... but i don't what to be disrespectful in any way.